stevegal/solis_control #1: More of a question than an issue!

viking2010, 2023-06-23:
Hi, 

I know I mentioned it before in the issue here https://github.com/hultenvp/solis-sensor/issues/278  but are you able to share the documentation for the API? Solis do provide the v1 api documentation at this location [](https://solis-service.solisinverters.com/en/support/solutions/articles/44002212561-api-access-soliscloud) but the support agent I am dealing with at the moment, whilst they are responding to my queries, doesn't seem to fully understand what it is I'm asking for. 

I know one of their Product Managers has previously said their API support isn't great.  If you were able to share the documentation, then I can try and help provide some further support with this script if needs be. At the very least, do some testing for you.

Cheers

stevegal, 2023-08-01:
I did get sent the documentation, but it says "confidential" - so not sure if I'm allowed to share it on. I have it working now and it regularly updates by charging times (and discharge times) on my battery which is all a really need. Was there any particular functionality you were after and I could look at the documentation.

stevegal, 2023-08-02:
cool, but it's the control api that this refers to. 
```SolisCloud Control API V1.1.1``` if you want to search for it. You can really mess with all the inverter settings and I guess really break it. The documentation itself isn't very helpful for example

the charge/discharge settings that I use in this api has 
cid = 103
function = Charge/Discharge Setting
value = [{"name":"Charge Current 1","value":"1","type":"1","unit":"A","min":0,"max":100}, {"name":"Discharge Current1","value":"1","type":"1","unit":"A","min":0,"max":100}, {"name":"Charge Time1","name1":"Start Time","name2":"End Time","value":"1","type":"0"},
{"name":"Discharge Time1","name1":"Start Time","name2":"End Time","value":"1","type":"0"},
{"name":"Charge Current2","value":"1","type":"1","unit":"A","min":0,"max":100}, {"name":"Discharge Current2","value":"1","type":"1","unit":"A","min":0,"max":100}, {"name":"Charge Time2","name1":"Start Time","name2":"End Time","value":"1","type":"0"},
{"name":"Discharge Time2","name1":"Start Time","name2":"End Time","value":"1","type":"0"},
{"name":"Charge Current3","value":"1","type":"1","unit":"A","min":0,"max":100}, {"name":"Discharge Current3","value":"1","type":"1","unit":"A","min":0,"max":100}, {"name":"Charge Time3","name1":"Start Time","name2":"End Time","value":"1","type":"0"},
{"name":"Discharge Time3","name1":"Start Time","name2":"End Time","value":"1","type":"0"}]
Note that the related parameters must be set together with a comma "," to separate the time periods. No overlap is allowed. The order is:
Charge Current1 Unit A Range 0-100A
Discharge Current1 Unit A Range 0-100A
Charge Time1Start Time Format hh:mm
Charge Time1End Time Format hh:mm
Discharge Time1Start Time Format hh:mm
Discharge Time1End Time Format hh:mm
Charge Current2 Unit A Range 0-100A
Discharge Current2 Unit A Range 0-100A
Charge Time2Start Time Format hh:mm
Charge Time2End Time Format hh:mm
Discharge Time2Start Time Format hh:mm
Discharge Time2End Time Format hh:mm
Charge Current3 Unit A Range 0-100A
Discharge Current3 Unit A Range 0-100A
Charge Time3Start Time Format hh:mm
Charge Time3End Time Format hh:mm
Discharge Time3Start Time Format hh:mm
Discharge Time3End Time Format hh:mm
Example: {"cid":"103","value":"100,100,09:44,09:44,09:44,09:44,100,100,09 :44,09:44,09:44,09:44,99,100,09:44,09:44,09:44,09:44"}

There is another section on getting your password in the right format which just gives minimal documentation, and you can see the results in the "login" api call in the code. It doesn't tell you what to do with the token you get back for the control api, but a bit of guesswork soon sorted that out.

stevegal, 2023-08-02:
also, I just raised a ticket once I activated the monitoring api saying something like
"I have access to read data from the api, but was wondering if it would be possible to have access to set charging times via the api. This would help me automate the amount of charge I would require given the solar forecast for the following day. I am a programmer, so have experience working with api's."

and then they responded with the control documentation. Maybe try that route?

stevegal, 2023-08-17:
@viking2010 - is it okay to close this now?

closed 2023-09-02 by stevegal