stevegal/solis_control #2: Time of use Switch

viking2010, 2023-07-26:
Hi

Can you advise if there is a way to enable/disable the time of use switch on the inverter from the API?

Many thanks

stevegal, 2023-08-01:
by setting the times as you have suggested (00:00 to 00:00) then this will turn that slot off. Note you have 3 slots that need to filled, each slot can have a discharge and charge time. This basically does the same thing as the menus on the solid inverter itself and after you have used it then you can check on the inverter

stevegal, 2023-08-01:
I use this in homeassistant to estimate how much I'll need to put into my battery based on solcast forecast for the day ahead (+ a correction factor as this always seems to over-predict), the current house usage (in case I need a boost to get me to when my panels start to charge) and my estimate for the following day. Not perfect, but it just runs as an automation and gets me most of the way there.

stevegal, 2023-08-07:
```
@service  
async def solis_control(demand):
    config = pyscript.config["apps"]["solis_control"]["config"]
    
    battery_kwh = float(state.get(config['battery_soc']['entityId']))*0.096
    min_battery_kwh = 9.6*0.20
    buffer_kwh = 0.25
    tomorrow_prediction = float(state.get(config['solar_tomorrow']['entityId'])) * 0.85
    house_power = min(0.5,float(state.get(config['house_power']['entityId']))/1000.0)
    
    car_plugged = state.get(config['charger_plug']['entityId']) == 'on'
    log.warning("is car plugged?"+str(car_plugged))
    
    now = datetime.now()
    next_charge_window = (now + timedelta(days=1)).replace(hour=4, minute=30, second=0)
    start_charge_window = (now + timedelta(days=1)).replace(hour=0, minute=30, second=0)
    solar_start =  (now + timedelta(days=1)).replace(hour=7, minute=20, second=0)
    
    hours_next_charge_window = (next_charge_window - now).total_seconds()/3600.0
    battery_load = battery_kwh - (hours_next_charge_window * house_power)
    solar_time = (solar_start - next_charge_window).total_seconds()/3600
    
    today_index = next_charge_window.weekday()
    
    battery_available_charge_kwh = min_battery_kwh if car_plugged else max(min_battery_kwh, battery_load)
    solar_load = solar_time * house_power 
    
    today_demand = demand[today_index]
    
    morning_shortfall_kwh = solar_load + min_battery_kwh + buffer_kwh - battery_available_charge_kwh 
    log.warning("today demand:"+str(today_demand)+ " min battery:"+ str(min_battery_kwh+buffer_kwh )+ " available:"+str(battery_available_charge_kwh)+ " tomorrow gen"+str(tomorrow_prediction))
    afternoon_shortfall = today_demand + min_battery_kwh + buffer_kwh - battery_available_charge_kwh - tomorrow_prediction
    
    log.warning("morning shortfall:"+ str(morning_shortfall_kwh) +" afternoon shortfall:"+ str(afternoon_shortfall))
    shortfall_kwh = max(morning_shortfall_kwh, afternoon_shortfall)
    times = []
    if (shortfall_kwh > 0):
      log.warning("shortfall:"+str(shortfall_kwh) + " (demand="+str(today_demand)+", house power:"+ str(house_power)+ ", battery: "+str(battery_kwh)+", "+str(battery_load)+", "+str(battery_available_charge_kwh)+")")
      charge_secs = shortfall_kwh*3600/2.5
      start_time = max(start_charge_window,next_charge_window - timedelta(seconds=charge_secs))
      log.warning("start charge at:"+start_time.strftime("%a, %d %b %Y %H:%M:%S"))
      times.append({"chargeCurrent":"50", "dischargeCurrent":"50", "chargeStartTime":start_time.strftime("%H:%M"), "chargeEndTime":next_charge_window.strftime("%H:%M"), "dischargeStartTime":"00:00", "dischargeEndTime":"00:00"})
      times.append({"chargeCurrent":"50", "dischargeCurrent":"50", "chargeStartTime":"00:00", "chargeEndTime":"00:00", "dischargeStartTime":"00:00", "dischargeEndTime":"00:00"})
      times.append({"chargeCurrent":"50", "dischargeCurrent":"50", "chargeStartTime":"00:00", "chargeEndTime":"00:00", "dischargeStartTime":"00:00", "dischargeEndTime":"00:00"})
    else:
      # clear any charge times that may be set
      log.warning("no shortfall - clearing")
      times.append({"chargeCurrent":"50", "dischargeCurrent":"50", "chargeStartTime":"00:00", "chargeEndTime":"00:00", "dischargeStartTime":"00:00", "dischargeEndTime":"00:00"})
      times.append({"chargeCurrent":"50", "dischargeCurrent":"50", "chargeStartTime":"00:00", "chargeEndTime":"00:00", "dischargeStartTime":"00:00", "dischargeEndTime":"00:00"})
      times.append({"chargeCurrent":"50", "dischargeCurrent":"50", "chargeStartTime":"00:00", "chargeEndTime":"00:00", "dischargeStartTime":"00:00", "dischargeEndTime":"00:00"})
    inverterId= getInverterList(config)
    token = login(config)
    set_control_times(token, inverterId, config, times)
```

then I just set it up to run at 23:00 each evening. The `0.096` is my battery size. You can see that I just have a set cheap window for octopus and only use one charging window, but that should give you an idea... and you can see I've been quite lazy with the "solar start"... I should probably get that from the prediction...

stevegal, 2023-08-07:
to get it working, just make suer you set all the configuration up in the main configuration.yaml for homeassistant in the pyscript section
```
# enable python script
pyscript:
  allow_all_imports: true
  hass_is_global: true
  apps:
    solis_control:
      config:
```

the demand part is simply passed in with the automation yaml. Hope that makes sense

stevegal, 2023-08-17:
@viking2010 - is it okay to close this issue now?

closed 2023-09-02 by stevegal