stevegal/solis_control #6: CID

edrocket, 2023-09-01:
Hi, great work, waiting for the control documentation myself, but in the mean time your scripting is excellent.
But, the CID changes, is my analysis.
"103" worked, until today. It seems to be "4692" (for feedback prio, for self use it is "5007".

Any thoughts on the CID definitions? 
I cannot figure our when/where this is defined (while monitoring the web communications)

stevegal, 2023-09-02:
Thanks for your heads up @edrocket.  I've just noticed that it's failed as well with an error message of `Send failed !Three elements of the equipment are not stored`. I'm guessing that they have changed the api in some way without notification.

I'll raise a ticket with solis again and ask for updated documentation. If you get a response before me, please let me know!

stevegal, 2023-09-02:
Not sure I'd just be gussing the CID numbers as you can do *a lot* to the inverter - so I'd not risk sending random values. I'm not sure what you mean about `4692` and `5007`. Did you just change the CID values by guessing?

stevegal, 2023-09-02:
I've just been to the ticket site on the https://solis-service.solisinverters.com/en/support site and submitted a ticket. I've just noticed that there is a new type of ticket for `End User Control Access`, which sounds just like what we need. I suppose that they have reworked how this works. I'll update this thread if I get any further information.

edrocket, 2023-09-02:
Hi Steve, 

I did not really guess the cid, it's used by the cloud app (sniffing the network packets with firefox). 

But the logic why it is changed lacks. 

Edwin

stevegal, 2023-09-02:
it looks like the api has changed version to v3...

Looking at the url in the control section of the side the url has changed to `https://www.soliscloud.com:15555/inverter/atCommandV3` and the payload has changed a bit as well. Looks like that battery charge cid has changed to `4644` in that version of the website for my inverter. There is also at `atRead` url which sounds interesting as well and we may be able to hook that up to a sensor.

I got this from the new inverter control section of the website. Is that where you got your information from?

stevegal, 2023-09-02:
Just tried that, but it also looks like the login for v3 has changed as well. I'll see if I can figure that out as well.

stevegal, 2023-09-02:
just tried to work with the v3 api, but, without the documentation or approval to have a new api secret and resource added, it looks like I cannot log in... However, just changing that v2 to the `4644` CID name seems to work just fine for me. I'll wait for some documentation as you seem to have different CID values than me and I'd hate for it to break someones inverter.

edrocket, 2023-09-03:
> it looks like the api has changed version to v3...
> 
> Looking at the url in the control section of the side the url has changed to `https://www.soliscloud.com:<port>/inverter/atCommandV3` and the payload has changed a bit as well. Looks like that battery charge cid has changed to `4644` in that version of the website for my inverter. There is also at `atRead` url which sounds interesting as well and we may be able to hook that up to a sensor.
> 
> I got this from the new inverter control section of the website. Is that where you got your information from? (inverter Control->Self Consumption -> Charge and Discharge1-3

I use Feedback Priority for now, a read request from the portal contains the CID I refer to, please see below

[image]

It is 5007 when I read the schedule in Self-Use.

stevegal, 2023-09-04:
interesting. So it looks like there a few options for seeing time of use charge and discharge. The one I was targetting was the "self consumption" one and not the "feedback priority". When I access feedback Priority "charge and Discharge1-3" I get a CID of `4658`!

When I was show around my inverter from the installer, they did show me the "self consumption" menu on the inverter and pointed me at that for home to set charging times for the battery.

I am a bit confused now as you have different CID values again. I wonder if this is model type specific? But I can't see where in the web app that is sourced from if true.

stevegal, 2023-09-04:
just found this endpoint... `https://www.soliscloud.com:15555/inverter/control/list` which is a POST endpoint... data payload
```{
  "InverterId": "your inverter id",
  "pid": "<some menu identifier>",
  "localTimeZone": 1,
  "language": "2"
}```

now this returns a list with id's, one of which is `4658` - the feedback one.... so I wonder if this is now different on each inverter model. Can you find this list endpoint and see if you can find the list for your menus. I've looked at each menu list endpoint and this is common with a different pid number on each menu. for example my feedback one seems to be 97 and self-consumption 96.

coathook, 2023-09-06:
hi
I also tried looking at the endpoint..
In "Self Consumption". when I change the timings for example (a small change)with stop discharge time changed from 5:30am to 5:28am. (slot 2)

For atCommandV3:

{"cid":"4644","inverterId":"xxxxxxxxxxxxx","value":"50,0,00:00,00:00,23:30,00:00,50,0,00:00,00:00,00:00,**05:28**,50,50,00:00,00:00,00:00,00:00","yuanzhi":"50,0,00:00-00:00,23:30-00:00,50,0,00:00-00:00,00:00-**05:30**,50,50,00:00-00:00,00:00-00:00","localTimeZone":1,"language":"2"}

(time changes in bold)

compared to the current coding, this seems to swap cid and inveterID values, and add "yuanzhi" (original values) plus "localTimeZone@ and "language". (maybe these are optional?)

does this help at all?
do you see the same?

stevegal, 2023-09-06:
yeah. I can't get the new v3 api endpoints to work... when I try to login I get a 401 response code....

thats the bad news..

BUT the good news is that we agree on the cid value... and... if you put that cid value into the v2 api endpoint it all seems to work. I suppose, at some time, they'll remove the v2 control endpoints, but it works with that value. I'll change it in the cid to `4644` in this for now. I'll try to find some time to reverse engineer the the login for the new control api. I'm not really getting anywhere with my requests for documentation to the new control api... they just keep sending me the monitoring api :(

stevegal, 2023-09-06:
@edrocket - updated... it should work now... until v2 api's are removed :(

stevegal, 2023-09-11:
@coathook - yeah, same thing. I've changed the cid's to match the new ones and left the control api endpoints alone which seems to work for now. I've not no luck with support as they keep pointing me at the monitoring api and not the control api. I guess I got lucky first time. I'll keep trying.

kris-g, 2023-09-12:
I've got the latest v2.0 documentation from support. It does seem its just some command IDs which have changed. The entire format of the document has changed, so hard to diff and find other potential changes.

ram1t, 2023-09-13:
> I've got the latest v2.0 documentation from support. It does seem its just some command IDs which have changed. The entire format of the document has changed, so hard to diff and find other potential changes.

v2.0 or v3.0?

kris-g, 2023-09-13:
The original document I think we were all sent was titled v1.1.1 and this new document is titled v2.0. But I'm not sure any of that has anything to do with the fact that some of the REST URLs used by the control mode web UI are suffixed v3, eg. atCommandV3. I have a feeling those URLs always had that suffix going back the last couple months.

stevegal, 2023-09-14:
@kris-g It's the v1.1.1 ones I have and this script is based on. I wish I had the v3 documentation to move it to the newer api as they'll probably deprecate the v1.1.1 ones through time. I don't suppose you could share the new sections of the documentation (especially the authentication section). Have you seen the section above where we are looking at the atCommandv3 url and body? Does that match the documentation?

kris-g, 2023-09-15:
I can send you the document privately. Don't really want to publish it publicly due to the NDA.

kris-g, 2023-09-20:
> @kris-g opened up a private repo and invited you. Push the pdf there and I'll delete the repo afterwards

Yep already done last night. 👍🏽

coathook, 2023-09-20:
My battery charge schedule failed last night, looking at the log the error message is back

Send failed !Three elements of the equipment are not stored.

I’ll check the code again later but anyone else seeing the same?  Cid  is 4644

stevegal, 2023-09-20:
Just tried... and yes, changd back to 103 (website as well). Changing the cid in the script back to 103 now makes it work. Wonder what they are doing?

coathook, 2023-09-22:
not sure what's going on with solis - CID (for self consumption) is back again to 4644 today ...
I'm thinking it would be good to display the response from the service to be sure the command has worked correctly...

TGRolando, 2023-10-25:
Hi @stevegal, 
Great work on this.
@damo642 has had some success with getting a python program running. May be worth a chat with him.
Cheers.

coathook, 2023-10-26:
HI all
just taken a look at the solis inverter control via the solis website, the UI has changed quite a bit, and there are now 6 charge and discharge slots
looking at the endpoint format there are some changes too.. here is an example

{"cid":"103","inverterId":"XXXXXXXX","value":"50,0,00:00-00:00,23:30-00:00,50,0,00:00-00:00,00:00-02:44,50,0,02:45-05:29,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00","yuanzhi":"50,0,00:00-00:00,23:30-00:00,50,0,00:00-00:00,00:00-02:59,50,0,03:00-05:29,00:00-00:00","localTimeZone":1,"language":"2"}

I have successfully managed to update your code to change the comma separation between start and stop times to a "-", and update to 6 slots..

stevegal, 2024-01-08:
this seems to have stabilised now. Closing