stevegal/solis_control #14: Templating values

chris-y, 2024-02-07:
This might be more of a HA or PyScript question but I'll try here first.

I'm trying to insert a charging rate into the config I provide to the script.  I'm attempting it like so:

```
service: pyscript.solis_control
metadata: {}
data:
  days:
    - chargeCurrent: '{{ sensor.solis_battery_charge_rate }}'
    [...]
```

Where sensor.solis_battery_charge_rate is a helper I've set up which contains the value I want to insert.

I'm getting this error when running:
`Error: Error rendering data template: UndefinedError: 'sensor' is undefined`

Is templating like this supported? (I figured HA would resolve the values and then pass it, so pyscript wouldn't even know)  Or so I have to do something more complicated such as in #13 ?
I've also tried creating a script in HA which calls the solis_control script and has a field with the value.  That doesn't seem to want to insert either - I just get blank data in that case.

btw, thank you for this script, it has been invaluable.  I usually just set it to run with a 24hr charge or discharge - which starts the inverter immediately - when something triggers "on" (like a saving session) and then tell it to reset back to my usual config when that trigger turns off.  With this I've not needed to send particular times to the script but I do want to tweak the charge current for my overnight charging so I can get the battery to a certain percentage rather than relying on hardcoding a few different currents and hoping for the best.

chris-y, 2024-02-09:
OK, I made a silly mistake there, the line should be something like:
```
   - chargeCurrent: "  {{ states('sensor.solis_battery_charge_rate') }} "
```

However, this turns up in the config without the quotes, so it still doesn't work.

Whatever I try, I can't get the quotes to attach properly, I either get too many or none!

Dixie81, 2024-02-09:
I've been setting the charging current using this script and a template in an automation. It took me a while to get something that worked but the code that eventually got the correct values was this:

```
- chargeCurrent: 0{{ states('sensor.estimated_charge_rate') | string }}
```

To be honest I can't remember the exact reasoning behind it all now but remember prepending the 0 and converting the sensor value to a string were both needed to get the correct value sent to the inverter.

chris-y, 2024-02-10:
Yes, the 0 seems to be the key part. With it the quotes get inserted (along with an extra leading zero, which the inverter or script don't appear to mind). All working now, thanks!

closed 2024-02-10 by chris-y