stevegal/solis_control #16: getInverterList v1 api isnt working anymore

aymsley, 2024-02-27:
Good morning, 

Firstly thank you so much for writing this. I've just got my PV installed and was expecting to have to use solarman only to find I had an S3 stick. Your interface with the API is awesome. 

I was playing with it last night and I've noticed your getInverterList call isnt working. 

2024-02-27 09:59:51.962 DEBUG (MainThread) [custom_components.pyscript.eval] file.solis_control.solis_control: calling post("https://www.soliscloud.com:13333/v1/api/inverterList", {'data': '{"stationId":"XXXXXX"}', 'headers': {'Content-MD5': 'XXXXXXXX==', 'Content-Type': 'application/json', 'Date': 'Tue, 27 Feb 2024 09:59:51 GMT', 'Authorization': 'API XXXXXX='}})
2024-02-27 09:59:52.199 DEBUG (MainThread) [custom_components.pyscript.eval] file.solis_control.solis_control: calling json(, {})
2024-02-27 09:59:52.201 ERROR (MainThread) [custom_components.pyscript.file.solis_control.solis_control] Exception in <file.solis_control.solis_control> line 92:
        for record in inverterList['data']['page']['records']:

Using Chrome tools I see the page is calling listV2 now. 

https://www.soliscloud.com:15555/inverter/listV2

The response looks good

{
    "success": true,
    "code": "0",
    "msg": "success",
    "data": {
        "page": {
            "records": [
                {
                    "id": "xxxxx",

I have tried to update the 

INVERTER_URL= '/inverter/listV2'

and port 15555 but it didnt work. 

    response = await session.post("https://www.soliscloud.com:15555"+INVERTER_URL, data = body, headers = header)

2024-02-27 10:11:54.840 DEBUG (MainThread) [custom_components.pyscript.eval] file.solis_control.solis_control: calling post("https://www.soliscloud.com:15555/inverter/listV2", {'data'...

2024-02-27 10:11:54.945 ERROR (MainThread) [custom_components.pyscript.file.solis_control.solis_control] Exception in <file.solis_control.solis_control> line 93:
        for record in inverterList['data']['page']['records']:
                                   ^
KeyError: 'data'

Are you already aware of this problem. If I manually configure the inverterId the rest of the script works. Is there any reason why you dont just pass it in as a config item unless you are expecting people to have more than 1 inverter? You can harvest the inverer Id from solis cloud. 

https://www.soliscloud.com/#/station/inverter/inverterdetail?id=INVERTERID&stationId=xxxx

eddysteurs, 2024-03-01:
I am nut sure but think I get the same error since yesterday (29 feb) (my chinese is not so good)
[image]

aymsley, 2024-03-08:
> Still working fine for me. Check your configuration for password and api
> key values in case you are getting rejected due to incorrect values in
> either of those.
>
> I haven't been able to get hold of any documentation for the other
> versions of the api, but I suspect the v2 ones have a slightly different
> authorization mechanism.

Strange. I’ve commented your inverter def and just pass my inverter id
directly and that’s working fine. Oh well doesn’t matter, what I have works
fine and is one less api call so all good.

bryanajwells, 2024-04-14:
I have the same issue:
Exception in <file.solis_control.solis_control> line 92:
for record in inverterList['data']['page']['records']:
^
KeyError: 'data'

I have checked and re-checked the config.
I too have commented out the inverter def block - thanks aymsley.

stevegal, 2024-04-17:
I've now had a few people with this error. I was wondering if you could provide some debug for me as it's working fine with my inverter. can you log the response of your solis inverterList by adding `log.warning(str(inverterList))` after line 90 where we make the post call and before I look for the inverterId in the response (line 92) and see if there is an error message there.

Thanks

stevegal, 2024-04-17:
can I close this issue?

aymsley, 2024-04-18:
Absolutely

stevegal, 2024-04-18:
see updated README for missing link as to where to find the plantId