stevegal/solis_control #17: new function request : other self-use mode parameters

eddysteurs, 2024-03-06:
Would itbe possible to make a seperate script to set the other parameters of "self-use mode" , this would allow more flexibility on how to set charge-discharge especially with summertime coming. I propose not to mix with the charge-discharge timing but in a second script.

[image]

stevegal, 2024-03-08:
unfortunately the api only exposes the charge and discharge times together. You could read the times then only update one of them I suppose which would give you the approximate functionality

eddysteurs, 2024-03-08:
Sorry that is not my question, the hours script works fine, but I would also like to automate the other parameters like "time of use switch"

aymsley, 2024-04-15:
There is a cid, 636 that should allow you to change the mode. According to the documentation you need to pass a 12 bit string. I've not had the chance to get it working yet. I've also asked for clarity as to what bit does what but haven't had much luck from the support team. The issue being you could end up changing other settings without realising. I did this in testing so stopped for now. 

BIT00 Self-generating mode switch 0-Off 1-ON
BIT01 Optimum yield mode switch 0-Off 1-ON
BIT02 Energy storage off-grid mode switch 0-Off 1-ON
BIT03 Battery wake-up switch (1-wake-up enabled 0-wake-up not enabled) 0-Off 1-ON
BIT04 Battery backup mode switch 0-Off 1-On
BIT05 Allow/disallow drawing power from the grid to charge the battery 0-disallow, 1-allow
BIT06 Feeder grid priority mode switch 0-Off 1-ON
BIT07 Nighttime battery over-discharge retention enable switch 0-Off 1-ON
BIT08 Dynamic adjustment of mains pick-up during strong battery charge enable switch 0-Off 1-ON
BIT09 Battery current correction enable switch 0-Off 1-On
BIT10 Battery treatment mode 0-Off 1-ON
BIT11 Peak-shaving mode switch 0-OFF 1-ON

aymsley, 2024-05-07:
@eddysteurs how about this? cid 636 as above wasnt working and Solis support couldnt help me. Despite the documentation said cid 142 is deprecated it still works for me fine. 

I watched this value for yuanzhi (previous value) whilst changing the mode via the web UI and observed it being eithr 866 or 803, if you dont pass it you get an error in chinese saying you must read the previous value. So I set it as below. I've been testing it out and it seems fine. Note I only use these two modes. I've not tried the "Off-Grid mode". 

    if mode=="Feed-in":
        value="1"
        yuanzhi="866" #means previous value in chinese
    else:
        value="0"
        yuanzhi = "803"

    #cid 142 is the feed in control. When 1 it's enabled. When 0 it's disabled and self-use is enabled.
    body = f'{{ \"inverterId\":\"{inverterId}\", \"cid\":142, \"value\":{value}, \"yuanzhi\":{yuanzhi} }}'

mikebatgithub, 2024-05-10:
Can I ask where the API documentation is of /v2/api/control functions? I have the /v1/api/xxxx documentation and have worked out, with the help of your python script, how to use /v2/api/login.  I'd like to have the control API documentation - do you have a link to it? Thanks.

aymsley, 2024-05-12:
You have to contact support and sign an NDA to get access to the control
api / documentation.