stevegal/solis_control #18: Multiple inverters

jonnyrider, 2024-04-14:
Hi,

Trying to get this to work, but I have 2 inverters (one controls the battery, the other receives the solar) and when I send the command I get confirmation that all has succeeded, but it is sending to the wrong inverter, so nothing actually happens.

I can't see anywhere to manually enter the inverter ID, and I've tried putting this in instead of the plant ID it doesn't work.

I can see in your code that the inverter ID is looked up from Solis, is there a way to hardcode the correct one in, or an option to change?

Thanks

Jonathan

jonnyrider, 2024-04-14:
Found where it needs to change, but this is as far as my python skills go!

```
def control_body(inverterId, chargeSettings) -> str:
    body = '{"inverterId":"**xxxxxxxxxxxxxxxx**", "cid":"103","value":"'
    for index, time in enumerate(chargeSettings):
        body = body +time['chargeCurrent']+","+time['dischargeCurrent']+","+time['chargeStartTime']+","+time['chargeEndTime']+","+time['dischargeStartTime']+","+time['dischargeEndTime']
        if (index !=2):
            body = body+","
    return body+'"}'
```

aymsley, 2024-04-15:
This is what I did as the lookup for me (only have one inverter) wasn't working. The inverterID doesn't change so you should be set.

jonnyrider, 2024-04-16:
That's great, thanks.

closed 2024-04-16 by jonnyrider