stevegal/solis_control #19: Error when calling

usbrit, 2024-04-17:
Hi, I'm getting the following error when calling the script:
This error originated from a custom integration.

Logger: custom_components.pyscript.file.solis_control.solis_control
Source: custom_components/pyscript/eval.py:493
integration: Pyscript Python scripting (documentation, issues)
First occurred: 20:46:18 (4 occurrences)
Last logged: 20:52:37

Exception in <file.solis_control.solis_control> line 92: for record in inverterList['data']['page']['records']: ^ TypeError: 'NoneType' object is not subscriptable

Any ideas on a solution would be appreciated.
Thanks.

aymsley, 2024-04-17:
It's the same issue I posted back in Feb. 

https://github.com/stevegal/solis_control/issues/16

My fix was to hardcode the Inverter ID and avoid this additional api call

stevegal, 2024-04-17:
I've had a few reports about this now but it works fine for me. I'll try to have a look over the weekend in case there is a difference between the version I'm using and the github version. So many people reporting the same issue can't be wrong!

stevegal, 2024-04-17:
can you add `log.warning(inverterList)` at between list 90 and 91

so it should look like
`
    response = await session.post("https://www.soliscloud.com:13333"+INVERTER_URL, data = body, headers = header)
    inverterList = response.json()
    inverterId = ""
`
this should log the the output string to the Homeassistant logs and we can see what response you got from the server

usbrit, 2024-04-17:
My error, used the wrong ID value, now working thanks!

bryanajwells, 2024-04-17:
> My error, used the wrong ID value, now working thanks!

Which ID did you get wrong and how?  I assume it was the plantID - maybe I'm making the same mistake.

usbrit, 2024-04-17:
@byranajwells, yes, I was using the Number listed one the web page, changed
it to the one in the address bar and it worked.

bryanajwells, 2024-04-17:
> can you add `log.warning(inverterList)` at between list 90 and 91
> 
> so it should look like
> 
> ```
>     response = await session.post("https://www.soliscloud.com:13333"+INVERTER_URL, data = body, headers = header)
>     inverterList = response.json()
>     log.warning(str(inverterList))
>     inverterId = ""
> ```
> 
> this should log the the output string to the Homeassistant logs and we can see what response you got from the server

I added the following two lines:
log.warning("solis response"+response.text())
log.warning(str(inverterList))

and these are the two results:
solis response:{"success":true,"code":"1","msg":"Communication error. Please refresh and try again later","data":null}
{'success': True, 'code': '1', 'msg': 'Communication error. Please refresh and try again later', 'data': None}

I also get the error:
Exception in <file.solis_control.solis_control> line 94: for record in inverterList['data']['page']['records']: ^ TypeError: 'NoneType' object is not subscriptable

line 94 is
for record in inverterList['data']['page']['records']:

Hope that helps.

bryanajwells, 2024-04-17:
> @byranajwells, yes, I was using the Number listed one the web page, changed it to the one in the address bar and it worked.
> […](#)

Ah yes - changed from the 6 digit PlantID to the much longer one in the URL and works perfectly.

stevegal, 2024-04-17:
ah yeah. It's documented how to find the plant-id in the solis integration which I link to. but I should improve that documentation

stevegal, 2024-04-17:
can I close this issue(updated readme with link)

usbrit, 2024-04-18:
Absolutely, thanks so much for your help. This is my "Holy Grail" moment for Solis!!

closed 2024-04-18 by stevegal