stevegal/solis_control #20: Time zone adjustment

usbrit, 2024-04-18:
Hi, Thanks to everyone's help here I've now got running automations controlling the inverter. Thank you all.

However, I'm getting a time sent t the inverter that is 1 hr after that provided by Bottlecapdave OctopusEnergy Target time.

I assume (as I'm running my HA system on GMT London, this a GMT vs BST issue.

Is there away to adjust this in the soils_control.py?  (maybe in time 33?:
'date = now.strftime("%a, %d %b %Y %H:%M:%S GMT")`

Thanks,
Usbrit

aymsley, 2024-04-19:
That time is used in the api protocol, not the chargetimes setting so
that’s not the problem.

Have you checked the time on the inverter? Mine was running in GMT, DST was
not enabled. There are two places to set it (confusingly). Having done this
I now see DST ON in the app.

usbrit, 2024-04-19:
Checked that, but appears to be a HA issue, not inverter.
Here's what the Target entity is outputting:
[image]
Start Time = 01.30

Here's the Trace from what HA is sending to the API:
[image]
Start Time = 02.30

Therefore, must be a HA issue?

aymsley, 2024-04-19:
I had to convert the time from the Octopus integration. You will see the time is +00 ie it's UTC time. You need to convert it if your inverter is following DST

aymsley, 2024-05-24:
Update: BottlecapDave did make a change and fix it. 

All attributes should now be displayed in local time in [v11.0.0](https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/releases/tag/v11.0.0). The majority of issues reported in this ticket were due to confusion in how the data was reported rather than it being wrong, so closing this ticket. If you still have an issue, please open a new ticket

Webreaper, 2025-02-04:
Wonder if I can ask a favour. I've built [my app](https://github.com/Webreaper/SolisAgileManager) based on the work in this repo (thanks for making it available and easy to follow). 

One question that's on my mind as we approach the end of March... I haven't put in any handling for Timezones in my API calls. It's not particularly clear from the Solis API docs whether I need to, or if I can just send and receive everything as UTC and it'll all just work. Any advice or insights (or decent docs) you can point me to?

aymsley, 2025-02-04:
I got stuck with this last year. There were posts on Reddit.

You can change it in the app for sure. I think there was a time setting for
the inverter too but it’s quite well hidden. I will see if I can find it.
But you can also email support for help.

As for the api. If the times on the inverter are correct ie it’s set to
follow DST then it will just take what you send it.

Check your app graph section.

See where it says, DST on.

Regards
Aymsley.

Webreaper, 2025-02-04:
I don't think you're understanding my question. I'm trying to establish what to pass in the API calls - can it just take UTC and it'll work everything out at the back end? 

So for example: 
- If I set a charge from 09:00UTC - 09:30UTC via the API the inverter will charge from 09:00 to 09:30. 
But: 
- If I set a charge from 09:00UTC - 09:30UTC via API in April, will it set the charge on the inverter from 10:00-10:30? Or will it set it to charge from 09:00DST - 09:30DST?

aymsley, 2025-02-04:
As I said. Your inverter will need to be set to follow DST. If you don’t it
will be GMT as, with mine at least, DST was not default “on”. So in April
you send 09:00. Your inverter thinks the time is 09:00 but the real time is
10:00 (gmt +1). So if you do nothing your charge times will be an hour
late.

Make sense?

Webreaper, 2025-02-04:
So are you saying that when I call the control API I should just pass the DST times (ie, UTC+1)?