stevegal/solis_control #21: Solis Cloud Returning JSON with Trailing Commas

CraigCallender, 2024-05-22:
FYI, I was getting an issue with the python script being able to execute the `login()`.  It turns out that solis cloud is returing invalid JSON as it contains trailing commas.  So I had to hack it like this:

```
async def login(config):
    body = '{"userInfo":"'+config['username']+'","password":"'+ passwordEncode(config['password'])+'"}'
    header = prepare_header(config, body, LOGIN_URL)
    response = await session.post("https://www.soliscloud.com:13333"+LOGIN_URL, data = body, headers = header)
    status = response.status
    r = json.loads(re.sub(r'("(?:\\?.)*?")|,\s*([]}])', r'\1\2', response.text()))
    if status == HTTPStatus.OK:
        result = r
    else:
        log.warning(status)
        result = response.text()
```

The invalid JSON looked like this (notice the trailing comma in the `data` object:
```
{
    "success": true,
    "code": "0",
    "data": {
        "token": "token_SOME_TOKEN",
    },
    "csrfToken": "token_SOME_TOKEN"
}
```

salvor-hardin, 2024-05-23:
I wrote my own php script and since yesterday it didn’t log in. Haven’t looked at the reason yet but we can bet is Solis returning invalid json for all clients, surprisingly their own app working.

the issue is not with our api clients but with Solis returning invalid json, they should end up fixing it as it doesn’t make sense; however this can be “prevented” by the client owners by “hacking” the response before interpreting as json, as you suggest @CraigCallender . Thanks for the tip, saved me time. 😅

aymsley, 2024-05-23:
Just checked and seeing the same thing.

aymsley, 2024-05-23:
Has anyone raised it with the support team?

CraigCallender, 2024-05-23:
I raised it with them already (ticket 376560) - but I'm sure other reports wouldn't hurt.  Here's what I sent so you can copy and paste:

> Hi Solis Support Team,
>   I'm raising an issue that seemed to have started in the last couple of days.  I'm using your v2 API to control my inverter and I've noticed that invalid JSON is being returned at the /v2/api/login endpoint.  The data object has a trailing comma - which isn't valid JSON and most JSON parsers abort trying to parse it.  Could you take a look and ensure other endpoints aren't doing the same?
> 
> I'm getting back data that looks like this:
> {
>     "success": true,
>     "code": "0",
>     "data": {
>         "token": "token_SOME_TOKEN",
>     },
>     "csrfToken": "token_SOME_TOKEN"
> }
> 
> Thanks!

stevegal, 2024-05-23:
I checked the logs from my solis inverted update from last night and they are clean. The inverted got updated with the times I asked for. So it's really strange that some are seeing this, but not all.

stevegal, 2024-05-23:
more playing... so yeah, I can now see this as well. Ideally I'd find a more lenient json parser for python and use that rather than the (rather neat!) attempt to find the trailing "," with a regex.

stevegal, 2024-05-24:
@CraigCallender - do you want to raise a pull request for teh trailing space regex so we all have something working?

salvor-hardin, 2024-05-24:
I changed my php script to replace ,} with } before attempting the json interpretation and it worked correctly overnight. Won’t harm you to add the replacement if they do the same again

CraigCallender, 2024-05-24:
@stevegal - Yep, I'll raise now.  But I agree we should find less of a hack long-term.  It looks like the JSON5 package in python will handle the trailing commas.  The caveat is that it's a "slow parser" - but considering the JSON we're parsing is so small I don't think that'll be a problem.

Solis Support came back to me and pretty much said, "Prove to us our Java examples don't work and we'll look into it."  Here's their full response:

> Firstly, refer to our most updated API document for your reference. If you encounter any issues after following that, please proceed with the steps below to rectify the API issue.
> 
> 1. Test your code using Postman API.
> 2. Kindly refer to our demo code, which is in JAVA and uses the Postman API exclusively.   
> 
>               [DEMO CODE IN JAVA](https://ginlong-product.oss-cn-shanghai.aliyuncs.com/templet/Authorization.java)
> 
> After following the above steps, if you still face issues, please provide us with the following details to help rectify your issue:
> 
> Please collect the complete request package, including request header, request body, and request frequency.

stevegal, 2024-05-24:
I'll merge that in now and move to the json5 this evening. I've tried it locally (the json5) and you can't really spot the speed difference - not that speed is very important. I'll update the read etc for the install as it's a bit more complicated.

@CraigCallender - thank you for the issue and (temp) fix!

closed 2024-05-24 by stevegal

lindseysuth, 2025-04-28:
@salvor-hardin greetings!  I have tried to connect to soliscloud to using your script, but, after 30,000 milliseconds it returns nothing - error message is that it could not connect to the server.  My credentials are working in postman and all of my headers seem to be identical in php and postman so now feeling a bit lost. Any further guidance much appreciated!  /L

PR #22 (2024-05-24): Issue 21: Fixing invalid JSON being returned by SolisCloud
Fixes #21.  Not a permanent fix as this can cause issues if `,}` is ever present within a JSON literal string.