stevegal/solis_control #28: Can't communicate with Solis

usbrit, 2024-10-15:
Getting the following error:

_This error originated from a custom integration.

Logger: custom_components.pyscript.file.solis_control.solis_control
Source: custom_components/pyscript/eval.py:1982
integration: Pyscript Python scripting (documentation, issues)
First occurred: 16:30:02 (2 occurrences)
Last logged: 17:11:36

solis response:{"code":"0","data":[{"msg":"Send failed !","code":1}],"time":"1729006201907"}

I've checked API details and they are correct. Using latest script.

Any suggestions?

jack00075, 2024-11-18:
Hi I  have got the same error.  Did you manage to resolve this?

stevegal, 2024-11-18:
I'm still getting a success on my inverter. I wonder if it's that cahnge in send code sneaking back in on some inverters?