stevegal/solis_control #32: Firmware 4B

fboundy, 2024-11-19:
HMI  Firmware 4B introduces a completely different way of controlling the charge slots. According to the latest code list document (https://oss.soliscloud.com/doc/SolisCloud_control_api_command_list.xlsx):

    New optimized earnings parameters. When the read parameter CID is 6798 and the result is 0xAA55, 
    use the new optimized earnings parameter.

The relevant codes for the individual entries seem to be those listed below.

I am not clear what `Volt n` controls as it doesn't appear on the web GUI. It is probably the minimum voltage at which to charge.

It's also not clear how the `Time Slot n` codes are set but my guess is that it will be using a string similar to the old version since they are reported back as `"00:00-00:00"`

Finally , by polling all the codes from 5916 to 5986 I have found that 5936 returns `"0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00"` which looks a lot like 103 did under the old system.

For reasons I won't bore you with my system is now offline until daylight but once it is back up and running I'll play around with this a bit more and investigate how easy it is to set these parameters.

Code | Value|Notice
-- | --- | --
5916 | Charge Time Slot 1 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5917 | Charge Time Slot 2 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5918 | Charge Time Slot 3 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5919 | Charge Time Slot 4 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5920 | Charge Time Slot 5 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5921 | Charge Time Slot 6 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5922 | Discharge Time Slot 1 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5923 | Discharge Time Slot 2 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5924 | Discharge Time Slot 3 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5925 | Discharge Time Slot 4 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5926 | Discharge Time Slot 5 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5927 | Discharge Time Slot 6 Switch | [{"name":"OFF","value":"0"},{"name":"ON","value":"1"}]
5946 | Charge Time Slot 1 |  
5928 | SOC1 |  
5947 | Volt 1 |  
5948 | Charge Current 1 |  
5949 | Charge Time Slot 2 |  
5929 | SOC2 |  
5950 | Volt 2 |  
5951 | Charge Current 2 |  
5952 | Charge Time Slot 3 |  
5930 | SOC3 |  
5953 | Volt 3 |  
5954 | Charge Current 3 |  
5955 | Charge Time Slot 4 |  
5931 | SOC4 |  
5956 | Volt 4 |  
5957 | Charge Current 4 |  
5958 | Charge Time Slot 5 |  
5932 | SOC5 |  
5959 | Volt 5 |  
5960 | Charge Current 5 |  
5961 | Charge Time Slot 6 |  
5933 | SOC6 |  
5962 | Volt 6 |  
5963 | Charge Current 6 |  
5964 | Discharge Time Slot 1 |  
5965 | SOC1 |  
5966 | Volt 1 |  
5967 | Discharge Current 1 |  
5968 | Discharge Time Slot 2 |  
5969 | SOC2 |  
5970 | Volt 2 |  
5971 | Discharge Current 2 |  
5972 | Discharge Time Slot 3 |  
5973 | SOC3 |  
5974 | Volt 3 |  
5975 | Discharge Current 3 |  
5976 | Discharge Time Slot 4 |  
5977 | SOC4 |  
5978 | Volt 4 |  
5979 | Discharge Current 4 |  
5980 | Discharge Time Slot 5 |  
5981 | SOC5 |  
5982 | Volt 5 |  
5983 | Discharge Current 5 |  
5987 | Discharge Time Slot 6 |  
5984 | SOC6 |  
5985 | Volt 6 |  
5986 | Discharge Current 6 |  
5955 | Charge Time Slot 4
5931 | SOC4
5956 | Volt 4
5957 | Charge Current 4
5958 | Charge Time Slot 5
5932 | SOC5
5959 | Volt 5
5960 | Charge Current 5
5961 | Charge Time Slot 6
5933 | SOC6
5962 | Volt 6
5963 | Charge Current 6
5964 | Discharge Time Slot 1
5965 | SOC1
5966 | Volt 1
5967 | Discharge Current 1
5968 | Discharge Time Slot 2
5969 | SOC2
5970 | Volt 2
5971 | Discharge Current 2
5972 | Discharge Time Slot 3
5973 | SOC3
5974 | Volt 3
5975 | Discharge Current 3
5976 | Discharge Time Slot 4
5977 | SOC4
5978 | Volt 4
5979 | Discharge Current 4
5980 | Discharge Time Slot 5
5981 | SOC5
5982 | Volt 5
5983 | Discharge Current 5
5987 | Discharge Time Slot 6
5984 | SOC6
5985 | Volt 6
5986 | Discharge Current 6

fboundy, 2024-11-20:
Full details are here: https://github.com/fboundy/pv_opt/issues/305#issuecomment-2489570654

I will endeavour to put together a PR to address this.

Webreaper, 2025-02-22:
Any progress with this? I'm looking to implement this in [my app](github.com/webreaper/solisagilemanager), and so if you've found anything useful that would be great.

Also, do you happen to know where in the API to find the firmware version so I can predicate the new logic rather than make it a manual setting?