stevegal/solis_control #34: Error in pyscript causing solis control to fail

abandyb, 2025-03-20:
from Home Assistant, when sending command to export:

Logger: custom_components.pyscript.file.solis_control.solis_control
Source: custom_components/pyscript/eval.py:511
integration: Pyscript Python scripting ([documentation](https://github.com/custom-components/pyscript), [issues](https://github.com/custom-components/pyscript/issues))
First occurred: 16:28:10 (1 occurrences)
Last logged: 16:28:10

Exception in <file.solis_control.solis_control> line 98: inverterList = response.json() ^ AttributeError: '_BaseRequestContextManager' object has no attribute 'json'

It is worth noting that I had an update (to 1.6.2) in HA this morning for pyscript

chrislewishome, 2025-03-21:
This looks to be a breaking change in version 1.6.2 of pyscript. I would downgrade back to 1.6.1 to get it working again, until someone with better python knowledge than me comes along and suggests a proper fix.

abandyb, 2025-03-21:
Yes, @chrislewishome, agreed!  -  that's what I ended up doing last night to get my automations working again. I had to restore from my previous backup prior to the update, as it didn't look possible to downgrade a HACS repo, so I lost a large portion of my day's statistics due to that

paulmthomas2, 2025-03-21:
I did manage to reinstall the previous pyscript version in HACS (so didn't
revert to back-up) which has fixed the issue for me currently too. But yes,
had same issues after the upgrade.

rustyscrew7, 2025-03-22:
Same problem here. Reverted to 1.6.1 :(

craigbarratt, 2025-03-22:
Sorry about the breaking change.  In 1.6.2 there were some changes around how `await` was handled.

Try removing the `await` on the previous line 116.

rustyscrew7, 2025-03-22:
> Sorry about the breaking change. In 1.6.2 there were some changes around how `await` was handled.
> 
> Try removing the `await` on the previous line 116.

Thanks for getting back @craigbarratt , sadly still the same issue though.

abandyb, 2025-03-23:
I have asked a question here [https://github.com/custom-components/pyscript/issues/710#issuecomment-2746109874] because there are 4 awaits in solis_control

craigbarratt, 2025-03-24:
I pushed a potential fix; please try the latest pyscript (master).

jparnellx, 2025-03-24:
> I pushed a potential fix; please try the latest pyscript (master).

Thanks
I downloaded and applied the latest version but still get the same issue

jparnellx, 2025-03-24:
> I pushed a potential fix; please try the latest pyscript (master).

I can confirm it's fixed following your changes.
Thanks

craigbarratt, 2025-03-24:
Released pyscript 1.6.3 with this fix.

rustyscrew7, 2025-03-25:
> Released pyscript 1.6.3 with this fix.

Thank you @craigbarratt